MarlinFirmware/AutoBuildMarlin #2: Install scripts

reloxx13, 2019-12-18:
Heya,

it would be nice to have install scripts per OS to copy the plugin to the desired default directory.

Would make updateing it easier.

thinkyhead, 2019-12-19:
It's not needed. You just use the built-in update features in **Visual Studio Code** to get the latest version, which is published on the [Extensions Marketplace](https://marketplace.visualstudio.com/items?itemName=MarlinFirmware.auto-build). On my computer VSCode is configured to automatically update all outdated extensions to the newest version when it launches, so there's no need to manually update it.

closed 2019-12-19 by thinkyhead

reloxx13, 2019-12-19:
Heya,
okay thanks for this info. I just never saw it updating.
E.g. for the new icons added sometime ago i had to update it myself.

Ill check my plugin update configuration and come back if it wont update :p