MarlinFirmware/AutoBuildMarlin #3: Fails to open correct type of terminal (needs a PlatformIO terminal)

sbaeder, 2019-12-19:
This is a reposting (so to speak) of the issue I had listed as #16095 in Marlin 2.0.

After a lot more playing around with this, I have determined that if I use a "normal" terminal as started from the vscode menu item for a "New terminal", that the command to run the "atom" script to do the build (buildroot\share\atom\auto_build.py), that it will fail.  On the other hand, if I use a platformio terminal, it works as expected.

I'm not sure how platformio changes the whole terminal, but the path and a lot of other env settings are present that are NOT a part of the default terminal.  So if you can make it invoke a platformio terminal, it will work.

This is on a clean install of both vscode and platformIO on Windows 10 as directed in the instructions.  
[image]

[image]

P.S. think splitting this out as an "extension" is great idea, and should probably include it's own copy of the build script...

brew99, 2019-12-19:
I can confirm the same issue is present with my Python 3.8 and Win10.   When using the Marlin autobuild "B" icon the same error is produced as when using the Terminal--> "New Terminal".

Unfortunately, I'm not a programmer, so am not much help in correcting anything like this.

thinkyhead, 2019-12-31:
Originally the extension opened up a Tk window with its own terminal environment. I'm not sure what other special steps were taken to ensure this window had the correct environment, but I'll take a look around. Meanwhile, grab an older version of **AutoBuildMarlin** from the Marlin repo that uses the Tk window (from around September) and see if it works.

sbaeder, 2020-01-01:
I'll look at that, but there is code to determine if it is inside VSC or in Atom, and I think the Tk window was for Atom (best I can tell from the actual script in the "atom" area that reads the config files, and does the right call to platformIO.  As stated, *IF* the right environment is established (as is done in the terminal opened by the platformIO command), things work as expected.  If it just tries to run the python code in the VSC window, it fails.

Will post after I try an older version, but it doesn't look like too many changes in the plugin or the script...

GMagician, 2020-01-04:
To solve this issue I added to system path, Win10, `C:\Users\[name]\.platformio\penv\Scripts`

sbaeder, 2020-01-04:
to me, that is a workaround - unless it is done by the normal installation of either platformIO or the plug-in.  Since the new terminal of platformIO can do it correctly, this plug in should too.

GMagician, 2020-01-04:
It's a workaround but I think problem has to be moved to vscode / platformio plugin.
I say that because when you open a terminal session and then try to launch 'pio' and this doesn't work, then this has nothing to do with "auto build"

sbaeder, 2020-01-04:
I'll disagree (slightly).  To me, since "auto build" is a vscode plugin AND PIOis a vscode plugin, it is all about how we work with the API's provided by both vscode AND PIO.  So it *is* an issue for auto build to handle.

As shown, using the "api" for a  "new terminal" command of vscode fails.  BUT, the PIO "new terminal" command creates a terminal that WORKS (without manually adding PIO to your path).  

So, if it (i.e. PIO) can do it, "auto build" should be able to as well.  In fact, PIO (as a vscode plugin) parses the entire platformio.ini file and creates "targets" for all the possible environments.  All we are trying to automate is the same thing, but based on parsing the configuration.h file and the various other files that map from the board name to the target environment.

The end goal (to me anyway) is a simple, easy to use system for people that may not really be "programmers", or care about things like "PATH" variables, etc...  If you already know what the target env is, it is easier to just use PIO and select the right target from it's menu.  Auto build is so you don't even have to know that.  You get the code, use a pre-made "config" for your machine, and click the button to build/load.

To be sure, this isn't a supre high priority bug, just something that can really help non-programmers update their 3d printers...

But as I said, I understand where you are coming from, and don't really disagree

Our goal for this plug-in (for visual studio) should be to do the work behind the scenes to figure out the actual target needed from the configuration.h file and invoke it so that it is "easier"...

GMagician, 2020-01-05:
@sbaeder from my point of view pio plugin should give to terminal access to its code, not marlin autobuild, even if latter could do on its own

GMagician, 2020-01-05:
@sbaeder have you tryed with `Update Terminal Path Configuration` in settings -> Extensions -> Platform IDE configuration ?

I played a little with it and now opening a terminal, it has pio program access

sbaeder, 2020-01-06:
I'll try that - if this works, then a simple addition to the installtion instructions to do this after you install PIO would be sufficient...

sbaeder, 2020-01-06:
Ran VSC and it wanted to update to 1.41.1, so I did that first.  Then in the VSC   File > Preferences > settings... (as above) ticked the box.  Same failure (i.e. makes no difference.   As shown before (maybe in the original ticket before this was split out), there is more different in a "normal" vsc terminal than the path to PIO - lots of other environment variables set too.

Just to answer @thinkyhead request from before - I tried the one from markin 1.9.x (hard to get older one without knowing more GIT stuff), and it opend a tk window, and just hung.  I had to delete the proper extension, and manually move folder into the .vscode area...when I deleted it, and reloaded vscode, it automatically had the most current version installed (it updated itself automatically ), which I think is pretty cool!

sbaeder, 2020-01-06:
new error messages are as follows

`Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Users\Owner\Documents\Marlin-2.0.0-VBaby> export TK_SILENCE_DEPRECATION=1 && [[ -f "buildroot/share/vscode/auto_build.py" ]] && python buildroot/share/vscode/auto_build.py build || python buildroot/share/atom/auto_build.py build
At line:1 char:33
+ export TK_SILENCE_DEPRECATION=1 && [[ -f "buildroot/share/vscode/auto ...
+                                 ~~
The token '&&' is not a valid statement separator in this version.
At line:1 char:37
+ export TK_SILENCE_DEPRECATION=1 && [[ -f "buildroot/share/vscode/auto ...  
+                                     ~
Missing type name after '['.
At line:1 char:84
+ ... ATION=1 && [[ -f "buildroot/share/vscode/auto_build.py" ]] && python   
...
+                                                                ~~
The token '&&' is not a valid statement separator in this version.
At line:1 char:137
+ ... y" ]] && python buildroot/share/vscode/auto_build.py build || python   
...
+                                                                ~~
The token '||' is not a valid statement separator in this version.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecor  
   dException
    + FullyQualifiedErrorId : InvalidEndOfLine
 
PS C:\Users\Owner\Documents\Marlin-2.0.0-VBaby> `

sbaeder, 2020-01-06:
So it looks like you are making some sort of progress...Let me know if I can help out.

GMagician, 2020-01-06:
@sbaeder:

> new error messages are as follows

that's the issue I have, please read #4 

In my PC I've solved, temporarily, replacing:
"C:\Users\[name]\.vscode\extensions\marlinfirmware.auto-build-2.0.1\extension.js"
with this file (please change extension when you download it)

[extension.txt](https://github.com/MarlinFirmware/AutoBuildMarlin/files/4025361/extension.txt)

thinkyhead, 2020-01-07:
Let me look into the `vscode.` and `workspace.` objects to see if I can detect the existence of the `auto_build.py` file by another means, rather than depending on the shell environment.

thinkyhead, 2020-01-08:
Ok, so the command line is fixed by using the node.js fs object.

It would be nice to eliminate the Python dependency, but we require some cross-platform method to present a dialog to the user. If there's something else that would be commonly installed on Windows that would be great.

Some kind of node.js plugin would be preferred over other options because we would have access to the dialog reply within the extension itself (which I'll look into) and then no external script may be needed at all once everything else is accomplished.

closed 2020-01-08 by thinkyhead

sbaeder, 2020-01-08:
@thinkyhead - Thanks for the efforts! Since Python is necessary for platformIO (and installed when PIO is installed as a plugin), I think we are OK with it as a dependency (for the popup that you have for when there are choices to make).

I'll look at this more and do some other testing, etc. But again...THANKS for all you do!

thinkyhead, 2020-01-08:
True that Python does need to be installed, and it is most common for it to have Tk included, so I won't sweat it too much. But I did find an alternative that should be able work within the VSCode window. See the https://github.com/anseki/vscode-color extension, which uses a small Electron app to make its dialog.