MarlinFirmware/AutoBuildMarlin #6: Still can't autobuild marlin on windows platform with default installation instructions.

sbaeder, 2020-01-17:
I am still unable to get the auto build code to work in a standard, basic installation of vscode and platformIO.   

1 - it can't find python, even though platformIO is written in python, and has a minimal installation included in the .platformio directory  (i.e. C:\Users\Owner\.platformio\penv\Scripts when "Owner is the username)

2 - even if I add that to the path, it fails to run properly, giving the same sort of errors I mentioned in issue #3 - since all of the platformio shell variables aren't set, it tries to open the tk window (i.e. doesn't recognize it is inside platformio) and then hangs with failure messages in the terminal window.

Steps to recreate

1. create a NEW account in windows to ensure a clean, default environment.
2. Install vscode and install platformIO following instructions - i.e. install vscode locally in the account, add platformio as an extension, and add the auto build marlin extension.
3. try to do a build. using the autobuild....fails
4. select the correct platform from the platformIO list - expand it, and select build.  WORKS.

This is a great idea, and the basic python script works fine to parse the files to pick out the configured boards build environment...But until auto build works, I would suggest that the instructions are changed so that they tell the user to look up the platform, and then use the platformIO command to do the build.

p.s. instructions also say to use the git clone command, and that doens't work either in a default installation.

sbaeder, 2020-01-17:
I can't believe it was this easy to "fix"....I decided to dig into it more, since it all works *IF* the environment is set right, and the platformio terminal command does that...

So looked more at all the platformIO code, and when they create a terminal, their code uses thier own methods that in an include file (the js code "terminal.js").  When it invokes the vscode method, it has a second parameter on the terminal creation to set the current platformio/vscode process environemnt into the new shells environment.

`var terminal = vscode.window.createTerminal({
      name: `Marlin ${title} #${NEXT_TERM_ID++}`,
      env: process.env
    });`

BINGO - works!  The new marlin build shell has the right environment, so it finds python, and the python code knows the environment and knows it is inside platformio, etc...

kreuzhofer, 2020-01-26:
I tried this fix and it only fixes part of the Problem. If I use your Version of Extension.js, I get a but further and then get this error:

PS C:\Users\daniel\Downloads\Marlin-2.0.x> python buildroot/share/atom/auto_build.py build
pwd: C:/Users/daniel/Downloads/Marlin-2.0.x
build_type: build
Working
python version 3.7.5
Traceback (most recent call last):
File "buildroot/share/atom/auto_build.py", line 994, in 
import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'
PS C:\Users\daniel\Downloads\Marlin-2.0.x>

kreuzhofer, 2020-01-26:
If i use the current Extension.js instead of your bugfix and install python for Windows, adding it to the PATH, it also starts but the error is different. So neither the bugfix nor installing python seperate fixes the current issue.

Windows PowerShell
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

Lernen Sie das neue plattformübergreifende PowerShell kennen – https://aka.ms/pscore6

PS C:\Users\daniel\Downloads\Marlin-2.0.x> python buildroot/share/atom/auto_build.py build
pwd:  C:/Users/daniel/Downloads/Marlin-2.0.x
build_type:   build 

Working

python version 3.8.1
build_type:   build
starting platformio
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python38-32\lib\threading.py", line 932, in _bootstrap_inner       
    self.run()
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python38-32\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "buildroot/share/atom/auto_build.py", line 1099, in <lambda>
    self.secondary_thread = threading.Thread(target=lambda q, arg1: q.put(run_PIO(arg1)), args=(que, ''))
  File "buildroot/share/atom/auto_build.py", line 906, in run_PIO
    pio_subprocess = subprocess.Popen(
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 1307, in _execute_child       
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden

kreuzhofer, 2020-01-26:
And with your bugfix AND python installed in the PATH the error is still 

ModuleNotFoundError: No module named 'tkinter'

So I suppose something has to be fixed with the python distro of platformio?

sbaeder, 2020-01-26:
Based on your error, it is more likely that the error in in the pythin code, and not the extension code to vscode (the .js file) that invokes the puthon script that parses things.

If a board needs additional configurations, the python code will try to create a pop-up box to get input.  Since my board doesn't need to ask for more input, it doesn't try to create the inteface window.

**Did you load the config.h file or if not, what board are you using,**

kreuzhofer, 2020-01-26:
The board is the Melzi from Ender 3. Seems that the python version that gets installed by platformio has no tcl/tk option installed

kreuzhofer, 2020-01-27:
Got to the ground of the issue. If you don't have Python installed before installing the platformio extension, it installs Python but with "tcl/tk" feature disabled. Normally this is enabled if you just click through the default installer. This leads to the issue of the "tkinter" error. However, your bugfix still seems to be needed to make it work together.

I suppose to make it work as a whole, your bugfix AND a bugfix to the platformIO installer needs to get published and a howto manual to fix the issue. My fixing instructions:

1 - download the file https://github.com/sbaeder/AutoBuildMarlin/raw/master/extension.js
2 - go to your user profile directory. This might be something like C:\Users\yourusername
3 - locate the subdirectory .vscode\extensions\marlinfirmware.auto-build-2.0.2
4 - overwrite the extension.js with the file you just downloaded
5 - Right click your windows start menu and open "Apps and Features"
6 - Klick into the search box which says "search this list"
7 - type "Python"
8 - If you see one or more Python installations click on the highest version number you find (for example 3.7.5) and then click "Modify", confirm the next question with yes
9 - The Python Installer opens, click again on "Modify"
10 - Make sure that the option checkbox for "tcl/tk…" is checked
11 - Click "next"
12 - Click "Install"
13 - If the installation bugs you with a popup window trying to find the installer exe like "python-3.7.5-amd64.exe", the installer has already been deleted by the system and you will have to fetch it from https://www.python.org/downloads/release/python-375/ for example and make sure you download the same installer that is requested by the installer popup from the list of files. For example, If it asks for a file ending with amd64.exe, select the x86-64 executable installer. Now point the installer popup to that downloaded file and it should finish the installation.
14 - You should now be able to use Auto Build Marlin - congratulations!

sbaeder, 2020-01-27:
I know @thinkyhead was thinking to use a javascript pop up eventually - might also be able to achieve same thing another way.  As I said, my board didn't need the tk window, so didn't need it installed...

Glad you got it going (and documented steps)!

mylife4aiurr, 2020-02-01:
Thank you @kreuzhofer and @sbaeder 
Could not use autobuild without the fixs here

dcs3473, 2020-02-01:
Can someone please explain how to do this fix on a MAC.  Almost same exact issue there.

thinkyhead, 2020-02-01:
Python comes with tk installed. If you have issues, try deleting the `.platformio/penv` folder and let platformio re-install it. That sometimes clears it up.

kreuzhofer, 2020-02-01:
@thinkyhead this might be true as soon as my last Bugfix makes it into the next platformio release. For now Python is installed by platformio with NO tcl support if you are on Windows. You have to go into the Apps and Features of Windows and modify the python installation to install tcltk. Not a big deal.

sbaeder, 2020-02-01:
@thinkyhead  - it is more than just the addition of tk to pythin.  You also have to pull in the change to the creation of the terminal by the javascript that is the extension.  The tk issue is for the .py script.

sbaeder, 2020-02-01:
@dcs3473 I would guess that on a mac, you would 1. add the line about the terminal to the .js file for the extension as the fist attempt.  Then if your board needs to open a tk window (and that fails), then add tk to the python installation used by platformio.

Note: without the change I made, it will cause python to fail to find the platformio and so that is first hurdle...

thinkyhead, 2020-02-11:
My aim is to completely replace the python script with Javascript and a web view, so stay tuned.

thinkyhead, 2020-02-15:
[image]

kreuzhofer, 2020-03-10:
I think we can close this issue. However, I am getting a new error since upgrading to latest 2.0.4.4: avr-g++: error: CreateProcess: No such file or directory. Not sure if this has to do with Auto Build Marlin or is another issue.

RockPaperPencil, 2020-03-28:
I encountered the "ModuleNotFoundError: No module named 'tkinter'" error on my Ubuntu system.
I resolved it by apt-installing the package "python3-tk". Hoping to help someone googling this :)

thinkyhead, 2020-03-31:
Well and good, but Tkinter is no longer used by this extension.

closed 2020-03-31 by thinkyhead