MarlinFirmware/AutoBuildMarlin #9: [2.0.3] Doesn't start!

GMagician, 2020-02-15:
Plugin has been updated just now and it seems it doesn't start anymore.
I can push buttons but nothing happens

arthurkok2, 2020-02-16:
Same here, the following error shows up in the extension host log

`[2020-02-16 12:03:57.422] [exthost] [error] TypeError: this.replace is not a function
	at String.toTitleCase (c:\Users\arthu\.vscode\extensions\marlinfirmware.auto-build-2.0.3\extension.js:3:52)
	at abm_command (c:\Users\arthu\.vscode\extensions\marlinfirmware.auto-build-2.0.3\extension.js:21:27)
	at c:\Users\arthu\.vscode\extensions\marlinfirmware.auto-build-2.0.3\extension.js:28:85
	at v._executeContributedCommand (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:543:163)
	at v.$executeContributedCommand (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:543:538)
	at p._doInvokeHandler (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:676:202)
	at p._invokeHandler (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:675:901)
	at p._receiveRequest (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:674:545)
	at p._receiveOneMessage (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:673:373)
	at c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:671:588
	at l.fire (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:48:541)
	at _.fire (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:190:934)
	at c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:838:649
	at l.fire (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:48:541)
	at _.fire (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:190:934)
	at t.PersistentProtocol._receiveMessage (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:195:277)
	at c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:192:459
	at l.fire (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:48:541)
	at p.acceptChunk (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:188:445)
	at c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:187:796
	at Socket.t (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:196:732)
	at Socket.emit (events.js:200:13)
	at addChunk (_stream_readable.js:294:12)
	at readableAddChunk (_stream_readable.js:275:11)
	at Socket.Readable.push (_stream_readable.js:210:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:166:17) mfbuild
[2020-02-16 12:04:00.289] [exthost] [error] TypeError: this.replace is not a function
	at String.toTitleCase (c:\Users\arthu\.vscode\extensions\marlinfirmware.auto-build-2.0.3\extension.js:3:52)
	at abm_command (c:\Users\arthu\.vscode\extensions\marlinfirmware.auto-build-2.0.3\extension.js:21:27)
	at c:\Users\arthu\.vscode\extensions\marlinfirmware.auto-build-2.0.3\extension.js:28:85
	at v._executeContributedCommand (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:543:163)
	at v.$executeContributedCommand (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:543:538)
	at p._doInvokeHandler (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:676:202)
	at p._invokeHandler (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:675:901)
	at p._receiveRequest (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:674:545)
	at p._receiveOneMessage (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:673:373)
	at c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:671:588
	at l.fire (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:48:541)
[... cut ...]
	at c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:187:796
	at Socket.t (c:\Users\arthu\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:196:732)
	at Socket.emit (events.js:200:13)
	at addChunk (_stream_readable.js:294:12)
	at readableAddChunk (_stream_readable.js:275:11)
	at Socket.Readable.push (_stream_readable.js:210:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:166:17) mfbuild
`

Taomyn, 2020-02-17:
I confirm the issue. I reverted the [Add toTitleCase string method](https://github.com/MarlinFirmware/AutoBuildMarlin/commit/3d69f8c9bb4156abeb3d71a7e733726eeb5e231f#diff-06f1274e40de25abda72d812b1cce86a) change and it works again.

GMagician, 2020-02-19:
This has been fixed by 2.1.0 version

closed 2020-02-19 by GMagician

Taomyn, 2020-02-20:
I like the new version, great job, and confirmed now working - at least it builds, can't test the final product in my printer yet.