MarlinFirmware/AutoBuildMarlin #11: [ISSUE] unknown MOTHBOARD displayed when attempting compile

NXSmiggy, 2020-03-28:
[image]

[image]

When attempting to build for BOARD_BTT_SKR_MINI_E3_V1_2 autobuild give the following error with no data populated. Reinstalll has been attempted, as has "Clean build" but no resoluton

thinkyhead, 2020-03-31:
It's working here, so I'm not sure what the issue could be.
[image]

McPrince96, 2020-04-23:
Same here but with a Rumba32 board. Tried other boards too, same result.

[image]

herbert-a, 2020-04-26:
Same here with skr 1.3
Using BOARD_BTT_V1_3

herbert-a, 2020-04-26:
I was able to solve the issue by copying the "#define MOTHERBOARD BOARD_RAMPS..." line from a fresh default copy of configuration.h into my version and then replacing the "RAMPS.." section with "BOARD_BTT_SKR_V1_3" hope this helps

thinkyhead, 2020-04-27:
Yep! When you want to define `MOTHERBOARD` …
it helps to start the line with `#define MOTHERBOARD`!

closed 2020-04-27 by thinkyhead