MarlinFirmware/AutoBuildMarlin #12: Auto Build Marlin build page blank

evilC, 2020-04-12:
After building quite happily yesterday, when I start my PC today, the build screen is blank - any ideas?
[image]

thinkyhead, 2020-04-13:
I've only seen that when the wrong folder was open, but the newest version puts up an alert to tell you if that's the case. If you want to see the exact error that's happening you have to use CTRL-SHIFT-P to bring up the magic box and then activate "Open Webview Developer Tools." However, the auto build panel needs to already be open, so it's possible the error will be missed.

evilC, 2020-04-14:
This is what I get when I open up the web view tools page you mentioned:
[image]

And here's a recording of VS in case it helps
[image]

thinkyhead, 2020-04-20:
If there's something in the config that it cannot parse then it might get stuck. Try it with unmodified default configurations and see if it works.

You can enable logging at the top of the `abm.js` file and then next time you restart the extension lots of messages will appear in the "Developer Tools" console — which you can find at the bottom of the Help menu.

McPrince96, 2020-04-23:
When i open an unmodified Marlin folder it works but not with a modified one. It only tells me "Use a specific environment for build" but the page isn't blank anymore.

I also commented on the other issue about the "Unknown motherboard" issue.
This issue  comes together with the "Unknown motherboard" issue.

evilC, 2020-04-24:
I can confirm that it is indeed to do with the folder. If I open a stock Marlin folder, the build screen is OK.
However, if I load my fork with my changes made, it is not OK.
This is odd, as I  swear what is in the repo is what I have already built and flashed it to my printer, so I don't know why it stopped working.

I am not sure I wholly understand how to get the debug info you requested
I found `abm.js` and set `const bugme = true`
Then, I am not sure how to "Restart the extension"?
Upon opening the developer tools windows (Ctrl+Shift+P), I see this:
[image]
Is that what you mean?

I can repro the issue by downloading my fork to a new folder and opening that folder in VS Code, the repo can be found here: https://github.com/evilC/Marlin/tree/Hemera

thinkyhead, 2020-04-27:
How is version 2.1.6 behaving?

evilC, 2020-04-27:
No change, same problem

thinkyhead, 2020-04-27:
ZIP your broken project and I'll try to see what ABM is choking on.

evilC, 2020-04-27:
https://github.com/evilC/Marlin/archive/Hemera.zip

n3bojs4, 2020-04-28:
I have exactly the same issue. I am running vscode 1.44.2 under linux.
the extension is version marlinfirmware.auto-build-2.1.7

n3bojs4, 2020-04-28:
This is more debug with 'code --verbose' :

`[18278:0428/102443.646064:INFO:CONSOLE(1611)] "%cTRACE color: #888 [text file model] load() - enter file:///home/MyUser/Documents/3D/firmwares/Marlin/Marlin/Configuration.h", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.715852:INFO:CONSOLE(1611)] "%cTRACE color: #888 CommandService#executeCommand abm.build", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.752582:INFO:CONSOLE(1611)] "%cTRACE color: #888 CommandService#executeCommand setContext", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.752893:INFO:CONSOLE(1611)] "%cTRACE color: #888 telemetry/editorOpened [object Object]", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.753200:INFO:CONSOLE(1611)] "%cTRACE color: #888 telemetry/webviews:createWebviewPanel [object Object]", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.769678:INFO:CONSOLE(1611)] "%cTRACE color: #888 [text file model] dispose() git:/home/MyUser/Documents/3D/firmwares/Marlin/Marlin/Configuration.h.git?{"path":"/home/MyUser/Documents/3D/firmwares/Marlin/Marlin/Configuration.h","ref":""}", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.769737:INFO:CONSOLE(1611)] "%cTRACE color: #888 [backup tracker] discarding backup git:/home/MyUser/Documents/3D/firmwares/Marlin/Marlin/Configuration.h.git?%7B%22path%22%3A%22%2Fhome%2FMyUser%2FDocuments%2F3D%2Ffirmwares%2FMarlin%2FMarlin%2FConfiguration.h%22%2C%22ref%22%3A%22%22%7D", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.840600:INFO:CONSOLE(1611)] "%cTRACE color: #888 IPty#resize 181 7", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.942632:INFO:CONSOLE(1611)] "%cTRACE color: #888 telemetry/webviewMissingCsp [object Object]", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102443.989689:INFO:CONSOLE(3291)] "Error: Unable to read file '/home/MyUser/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/js/jquery.jsonbrowser.js' (EntryNotFound (FileSystemError): Error: ENOENT: no such file or directory, open '/home/MyUser/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/js/jquery.jsonbrowser.js')", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (3291)
[18278:0428/102443.990436:INFO:CONSOLE(3291)] "Error: Unable to read file '/home/MyUser/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/css/jquery.jsonbrowser.css' (EntryNotFound (FileSystemError): Error: ENOENT: no such file or directory, open '/home/MyUser/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/css/jquery.jsonbrowser.css')", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (3291)
[18278:0428/102444.263347:INFO:CONSOLE(1611)] "%cTRACE color: #888 CommandService#executeCommand setContext", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102444.265926:INFO:CONSOLE(1611)] "%cTRACE color: #888 CommandService#executeCommand setContext", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (1611)
[18278:0428/102444.266243:INFO:CONSOLE(3983)] "[[object Object]]Cannot read property '1' of null", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (3983)
[18278:0428/102444.266275:INFO:CONSOLE(3983)] "TypeError: Cannot read property '1' of null
        at extractTempSensors (/home/MyUser/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/abm.js:375:29)
        at allFilesAreLoaded (/home/MyUser/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/abm.js:542:21)
        at /home/MyUser/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/abm.js:621:38
        at FSReqCallback.readFileAfterClose [as oncomplete] (internal/fs/read_file_context.js:61:3)", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js (3983)

`

thinkyhead, 2020-05-04:
Ok. I've just published 2.1.8 to fix the problem: It contained a stale reference to a removed JS file.

closed 2020-05-04 by thinkyhead

evilC, 2020-05-04:
Thanks for the fix, but unfortunately it does not fix the issue for me
[image]  
This appears to be an auto-generated file, if I edit the JSON and restart, the issue just comes back

n3bojs4, 2020-05-05:
I deleted files ~/.vscode/extensions/marlinfirmware.auto-build-* 
After restarting vscode, and reinstall the extension i had the same issue.
I added a new marlin freshly cloned directory to work on it, and now it's working. I don't understand because it's good for the old one too.
[image]

evilC, 2020-05-05:
> I added a new marlin freshly cloned directory to work on it,
For me, the issue does not happen with vanilla marlin, it only happens with my fork (https://github.com/evilC/Marlin/tree/Hemera ), but as I mentioned before, it built OK in the past, so no idea why it does not build now.  
It repros for me if I clone that branch of that repo to a fresh folder
FYI if you are having problems reproing this issue, I am in the Marlin Discord channel, am happy to let people remote in and debug

thinkyhead, 2020-05-06:
VSCode might aggressively cache some extension files, so it may help to:

- Uninstall the extension
- Quit VSCode / Restart VSCode
- Re-install the extension

The icon is usually blank on the first install, so you can:
- Quit VSCode / Restart VSCode, or…
- Install the "Reload" extension and use its button to reload the workspace.

thinkyhead, 2020-05-06:
@evilC — Here's a fix for you. I'll make ABM smarter about how it finds the thermistor list in the next release.
```diff
 *   999 : Dummy Table that ALWAYS reads 100°C or the temperature defined below.
 */
- // evilC change
- // Hemera uses different Temp Sensor to stock, so override
- //#define TEMP_SENSOR_0 1
- #define TEMP_SENSOR_0 5
+ #define TEMP_SENSOR_0 5 // evilC: Hemera uses different Temp Sensor (5 rather than 1)
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
```

evilC, 2020-05-06:
Most odd. I made that change and boom, it starts working again, but interestingly, I can change it back to how it was and it still works.
I suppose it's entirely possible that I added those comments after I actually built and uploaded, but obviously I did not think that adding comments would break the ability to build.
Whatever, thanks for finding the source of the problem!