MarlinFirmware/AutoBuildMarlin #13: Clicking build fails to start a new build

grokasm, 2020-04-30:
I am attempting to build the latest Marlin 2.0.x master branch and I just installed the latest ABM (2.1.7) and PlatformIO (1.10.0) extensions to vscode.
My first attempt at a build does not initiate the build process.
No visible UI changes occur on click as shown in the gif below:

[image]
On click, the following stack trace and exception show up in the developer console:

```
mainThreadExtensionService.ts:65 [[object Object]]e.with is not a function
$onExtensionRuntimeError @ mainThreadExtensionService.ts:65
_doInvokeHandler @ rpcProtocol.ts:398
_invokeHandler @ rpcProtocol.ts:383
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
(anonymous) @ rpcProtocol.ts:101
fire @ event.ts:581
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
(anonymous) @ ipc.net.ts:592
fire @ event.ts:581
acceptChunk @ ipc.net.ts:239
(anonymous) @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:200
addChunk @ _stream_readable.js:294
readableAddChunk @ _stream_readable.js:275
Readable.push @ _stream_readable.js:210
onStreamRead @ internal/stream_base_commons.js:166
mainThreadExtensionService.ts:66 TypeError: e.with is not a function
	at k.getFolder (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:357)
	at t.Configuration.getConsolidatedConfigurationModelForResource (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:346)
	at t.Configuration.getConsolidateConfigurationModel (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:345)
	at t.Configuration.getValue (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:342)
	at x.getConfiguration (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:686)
	at Object.getConfiguration (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:773)
	at terminal_command (/Users/dima/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/abm.js:785)
	at pio_command (/Users/dima/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/abm.js:839)
	at handleWebViewMessage (/Users/dima/.vscode/extensions/marlinfirmware.auto-build-2.1.7/abm/abm.js:1008)
	at l.fire (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:46)
	at p.$onMessage (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:650)
	at p._doInvokeHandler (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:663)
	at p._invokeHandler (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:662)
	at p._receiveRequest (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:661)
	at p._receiveOneMessage (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:660)
	at /private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:658
	at l.fire (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:46)
	at _.fire (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:189)
	at /private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:820
	at l.fire (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:46)
	at _.fire (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:189)
	at t.PersistentProtocol._receiveMessage (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:193)
	at /private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:190
	at l.fire (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:46)
	at p.acceptChunk (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:186)
	at /private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:186
	at Socket.t (/private/var/folders/91/q9d4bfq964ld16mrwvgsc84r0000gn/T/AppTranslocation/3082C5AD-8BAA-4C51-9A87-6C4439ACAD47/d/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:195)
	at Socket.emit (events.js:200)
	at addChunk (_stream_readable.js:294)
	at readableAddChunk (_stream_readable.js:275)
	at Socket.Readable.push (_stream_readable.js:210)
	at Pipe.onStreamRead (internal/stream_base_commons.js:166)
```

This may not be an ABM issue, any idea what may be causing this?

grokasm, 2020-04-30:
I had previously opened the project folder in vscode. I attempted to remove the folder and import the project via platformio and I no longer see that error on click, but the developer console does show the following: 

[image]

```
[Extension Host] runCommand platformio (8) ["-f", "-c", "vscode", "init", "--ide", "vscode", "--project-dir", "/Users/dima/dev/3dprinting/Marlin-2.0.x"] {}
3webviewElement.ts:241 [Embedded Page] WebSocket connection to 'ws://127.0.0.1:8010/wsrpc' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
console.ts:137 [Extension Host] runCommand /Users/dima/.platformio/penv/bin/python3 ["/Users/dima/.platformio/.cache/check-python.py"] {}
console.ts:137 [Extension Host] /Users/dima/.platformio/penv/bin/python3
console.ts:137 [Extension Host] runCommand /Users/dima/.platformio/penv/bin/python3 (2) ["-c", "import os;from platformio.project import helpers;l…oin(libdeps_dir, d)));print(":".join(watch_dirs))"]0: "-c"1: "import os;from platformio.project import helpers;libdeps_dir = helpers.get_project_libdeps_dir();watch_dirs = [helpers.get_project_global_lib_dir(), helpers.get_project_lib_dir(), libdeps_dir];watch_dirs.extend(os.path.join(libdeps_dir, d) for d in (os.listdir(libdeps_dir) if os.path.isdir(libdeps_dir) else []) if os.path.isdir(os.path.join(libdeps_dir, d)));print(":".join(watch_dirs))"length: 2__proto__: Array(0) {spawnOptions: {…}}
45webviewElement.ts:241 [Embedded Page] WebSocket connection to 'ws://127.0.0.1:8010/wsrpc' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
```
Perhaps the websocker error is related to the UI not updating, it seems that the process was spawned correctly...

thinkyhead, 2020-05-04:
Please test 2.1.8 to see if it works more reliably.

closed 2020-06-15 by thinkyhead