MarlinFirmware/AutoBuildMarlin #14: Refuses to see a process has completed.

drpain, 2020-05-24:
I can see it building successfully by keeping an eye on the console output, and echoing something to a file seemingly to acknowledged that the build / upload is done. But it just sits there for eternity.

And I have to restart VSCode which is incredibly annoying when you trying to make rapid changes. 

[image]
[image]

dtoebe, 2020-06-03:
Same issue

thinkyhead, 2020-06-15:
Weird. The capability for watching a file is provided at the high level by node.fs and it should be "trustworthy." ABM simply tells the system to start watching and then the system is supposed to report back when the file "changes" in any way. Perhaps if the contents are randomized instead of just writing "done" to the file then the system will be more likely to consider it "changed."

michaelhallnw, 2020-09-25:
This issue occurs for me in Windows when the VS Code default shell is set to something other than "cmd.exe" or "powershell.exe" (such as wsl.exe or bash.exe).

closed 2021-02-04 by thinkyhead

joakimbo, 2021-03-26:
I recently started tinkering with ABM and i too encountered this on windows, I was using git-bash as a default shell which cannot handle the windows filepath (backslashes).
Adding single quotation marks around the ipc_file path resolves this but instead breaks compatibility with cmd.
`if (ping) t.sendText(`echo "done" >${ipc_file}`);` -> `if (ping) t.sendText(`echo "done" >'${ipc_file}'`);`
It would be nice to have support for any integrated shell when running the commands in vscode, would the use of fs be better than to echo to the file?

PR #43 (2021-04-15): Fixes compatibility issue with (git) bash on windows
- Fixes #14, the problem when using (git) bash as shell in windows and shell incorrectly interprets ipc_file path, might not be the best way but it seems to do the trick