MarlinFirmware/AutoBuildMarlin #16: Marlin project folder not detected

potens1, 2020-06-17:
Hi,

I don't know if I'm not doing well the thing but I'm unable to make the marlin folder detected. I see an error about header folder (include) not detected, but the folder is not either in the Marlin sources.
[image]

Is there something I do wrong or a known problem ?

Platform: Linux
Vscode: 1.46
PlateformIO IDE: 1.10.0 
Autobuild Marlin: 2.1.15

potens1, 2020-06-17:
Ok, using marlin v2.0.0, it seems to work as expected, si, is the extension not compatible with 1.1.9 ?

thinkyhead, 2020-06-17:
Correct. Auto Build is not yet supporting Marlin 1.x.

closed 2020-06-17 by thinkyhead