MarlinFirmware/AutoBuildMarlin #19: Save build as .hex file

flub9, 2020-08-30:
There is probably some obvious method that I'm missing, however, I'm trying to upload and long story short, I need the compiled hex itself, instead of being able to use the built in upload feature. Is there any way I can do this? If not, I'd say that's a solid feature request.

Thanks for y'alls time and contributions.

thisiskeithb, 2020-09-19:
The compiled firmware location, be it *.hex or *.bin, isn't changed by using Auto Build Marlin.

Once you launch ABM and click Build, your compiled binary will be located in the `/.pio/build/%pio_environment_name%/` folder.

Example: If I'm compiling for an Ender-3 under the `melzi` environment, ABM builds the firmware and it can be found under `/.pio/build/melzi/firmware.hex`.

thinkyhead, 2020-09-24:
After doing the build with "Auto Build Marlin" you can click on the little folder icon to reveal the built firmware file.

closed 2020-09-24 by thinkyhead