MarlinFirmware/AutoBuildMarlin #20: Folder selected, but it says "The currently open folder doesnt contain Marlin project"

jasc76, 2020-10-09:
[image]

This is a valid folder, from my point of view

thinkyhead, 2020-10-18:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sakedas, 2021-01-06:
I have the same problem on Marlin 1.1.9.1. Seems that the extension doesn't like the folder structure of older Marlin versions

thinkyhead, 2021-02-04:
@sakedas — Marlin versions before 2.0 are not currently supported, but at some point I may get it working with 1.1.9.

closed 2021-02-04 by thinkyhead

66Drew66, 2023-11-18:
I have the same problem.
When I was having problems installing python it recognised the folder, now I have sorted that problem it's giving me this error message.