MarlinFirmware/AutoBuildMarlin #22: I compiled 2 firmwares perfectly, one for ender 3 one for a rambo board, and then i start getting this....

dissocio0, 2020-10-20:
I tried on 3 different computers, marlin auto compile spits out the same lines, that it does not recognize platformio or doesn't know where to find it. I find it odd that it worked twice before ... then I moved on to another computer and it didn't work from the start, then a third and again .. same error from the start. I know it must be an easy fix, and that somebody probably posted this, and someone is going to facepalm when seeing my post, but please tell me how to fix it, I really like the idea of auto-building marlin and its getting annoying not knowing why it's not working. I was able to compile the firmware through platformio no problem, flash it etc ... but I want to use auto build marlin (it even recognises the board and the firmware version, author and everything, just then I hit the hammer it spits out this error which I am tired of seeing)

error:
platformio : The term 'platformio' is not recognized as the name of a 
cmdlet, function, script file, or operable program. Check the spelling 
of the name, or if a path was included, verify that the path is correct 
and try again.
At line:1 char:1
+ platformio run -e rambo
+ ~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (platformio:String) [], Co
   mmandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Thanks for helping out if someone will help out, Cheers !

ivankravets, 2020-10-21:
@thinkyhead sorry for this issue. It's related to the recent update to PlatformIO IDE. We don't expose `platformio` CLI globally into the VSCode environment. It leads to multiple issues and brings conflicts for other extensions that use their own python environments.

See a workaround https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks/project.js#L144:L147

P.S: Do we need this `AutoBuildMarlin` extension? I hope the latest PlatformIO IDE 2.0 has all functionality for this? Developers can now switch between dev-environments from VSCode.

NormanDunbar, 2020-10-21:
Try adding the PlatformIO IDE commands to the path as per https://docs.platformio.org/en/latest/core/installation.html#windows -- this seems to work. See full thread at https://community.platformio.org/t/i-need-help-for-a-coding-noob/16621.

It looks like the auto build spawns a terminal session and runs pio there.

Cheers,
Norm.

thinkyhead, 2020-10-21:
I'll post a patch for Auto Build Marlin later today.

The ABM extension is quite valuable, as it saves the user the hassle of figuring out the exact PIO environment for their configured MOTHERBOARD, shows existing builds, reveals the built firmware file, etc. Plus, ABM will soon be including tools for downloading, editing, and migrating configurations.

When we start configuring Marlin exclusively with INI files, then we can talk about replacing ABM with PIO.

ivankravets, 2020-10-21:
@thinkyhead thanks for the quick response. Please use this hook, it will help https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks/project.js#L144:L147 

You need to pass this patched environment to your terminal.

thinkyhead, 2020-10-21:
Ah yes! I remember seeing that the terminal can get an environment passed in, and glad there's a hook for this issue. Is there anything I should check so ABM can maintain backward-compatibility with previous PIO?

thinkyhead, 2020-10-21:
@dissocio0 — Please update your Auto Build Marlin extension to version 2.1.22 and see if it fixes the issue for you.

ivankravets, 2020-10-22:
Terminal should work if you click on the terminal icon in the bottom blue bar. Also, you can use left side bar - PlatformIO - Quick Access - Open PlatformIO CLI. 

We don’t propagate PlatformIO in to the née default VSCode terminal. As I mentioned before, not all developers use PlatformIO just for embedded development. As result, the previous version of our extension makes problem for them.

thinkyhead, 2020-10-23:
I agree that an extension should not make global changes affecting all terminal sessions, and I don't mind that an extension like ABM (i.e., depending on another extension) should be required to use that other extension's methods when doing things like this. As I learn the process of making a VSCode extension you can be sure I'm taking down all these things in my notes!

closed 2020-10-23 by thinkyhead

dissocio0, 2020-10-24:
wow, lots of responses, thanks the patch seems to work... I just added the highlighted text in the terminal and it worked. Thank you!