MarlinFirmware/AutoBuildMarlin #23: Clean install of VsCode, PlatformIO IDE and Auto Build Marlin results if failed build on Mac OSX

madwax, 2020-10-22:
On Mac OSX 10.15.6
1) Clean install of vsCode and installing both PlatformIO IDE and Auto Build Marlin.
2) do git clone ... etc etc
3) Just do a mega2560 build 
It fails!
term says:
platformio run -e mega2560 ; echo "done" >/var/folders/04/82j6j0g16cdblk9w8_2lhq_h0000gn/T/ipc
XXXXXX@Alta Marlin % platformio run -e mega2560 ; echo "done" >/var/folders/04/82j6j0g16cdblk9w8_2lhq_h0000gn/T/ipc
zsh: command not found: platformio
XXXXX@Alta Marlin %

madwax, 2020-10-22:
Arr @thinkyhead You just fixed it in master!
2d252b9 - Use of PLATFORMIO_PATH as PATH works! Do you want me to close this?

thinkyhead, 2020-10-23:
I'm thrilled to hear it works! Closing…

closed 2020-10-23 by thinkyhead