MarlinFirmware/AutoBuildMarlin #45: Compiling with Autobuild Marlin Fails

sarvenn, 2021-06-20:
Whenever I try to build with autobuild marlin choosing RCT_BTT for my skr mini e3 v2 fail with the following error.
I can build firmware using platform.ini and selecting this environment from lower bar and clicking on build tick mark.
[image]
[image]

thisiskeithb, 2021-06-20:
> Whenever I try to build with autobuild marlin choosing RCT_BTT for my skr mini e3 v2 fail with the following error.

`RCT_BTT` is not a valid environment. You'll use one of the following with 2.0.9/latest bugfix-2.0.x:
- `STM32F103RC_btt`
- `STM32F103RC_btt_USB`
- `STM32F103RE_btt`
- `STM32F103RE_btt_USB`
- `STM32F103RC_btt_maple`
- `STM32F103RC_btt_USB_maple`
- `STM32F103RE_btt_maple`
- `STM32F103RE_btt_USB_maple`

> I can build firmware using platform.ini and selecting this environment from lower bar and clicking on build tick mark.

Neither of your screenshots have errors and using both Auto Build Marlin & manually editing `platformio.ini`, I can build for an SKR Mini E3 V2 in each environment without an issue.

Try deleting your hidden `.pio` folder within the Marlin project.

sarvenn, 2021-06-21:
I meant STM32F103RC_btt this one. My fault that I wrote wrong here.
I checked my board and my chip was RC not RE. In the past I was choosing the 512K version but after checking my chip code I choose the RC version yesterday.

I also deleted .pio folder and tried again with autobuild marlin.

There were two issue. One of them can be seen in the first screenshot=> multiple u8glib library.
Second is firmware build was unsuccessful due to missing file directory. sorry I missed to take and add the snapshot for that.

Second screenshot shows that I clicked on the environment (STM32F103RC_btt left side panel) in platformio extension and selected build option. That was successful.

I'm not at home now, so I'm not able to provide that screenshot now. I will do it tonight and upload here.

Besides that please notice that autobuild marlin and platformio build tast terminal lines are not the same. I don't know if they should match or not.

thinkyhead, 2021-07-17:
So, Marlin builds okay when you use the same `platformio …` command from the command-line, but not when it is run by the plugin?

sarvenn, 2021-07-18:
> So, Marlin builds okay when you use the same `platformio …` command from the command-line, but not when it is run by the plugin?

Sorry for late reply Scott. That's right. However I'm not sure it if is related with my computer or vscode. It was working previously. Last week I updated my marlin with platformio build task option and did not try autobuildmarlin plugin.
I'm at vacation now. When I turn back, I'll try and inform you.
Does anyone else faced the same problem? Do you know any case like this?