MarlinFirmware/AutoBuildMarlin #46: Build success but always recieve message 'Access to path is denied'

s-marley, 2021-07-18:
Once the build is successful I get the following output:

[image]

I think the issue might be that there is a space in the name of my home directory so it is truncating the path to C:\Users\Scott which of course isn't a real path. Perhaps missing some quotes somewhere?

Paul3899, 2021-08-24:
> Once the build is successful I get the following output:
> 
> 
> 
> [image]
> 
> 
> 
> I think the issue might be that there is a space in the name of my home directory so it is truncating the path to C:\Users\Scott which of course isn't a real path. Perhaps missing some quotes somewhere?

jbryson3, 2022-03-23:
I am getting a similar error on MacOS using Fish Shell. If its a different root cause, I can open up a new bug.

[image]

thinkyhead, 2022-05-21:
> I am getting a similar error on MacOS using Fish Shell. 

I have some new rough logic in the works, hopefully about correct, but maybe you can verify…
```js
const sh = vscode.env.shell,
      isbash = sh.indexOf('bash') >= 0,
      iszsh = sh.indexOf('zsh') >= 0,
      iscmd = sh.indexOf('cmd') >= 0,
      ispwsh = sh.search(/(powersh|pwsh)/i) >= 0,
      iswin = process.platform == 'win32',
      p = iscmd ? '&' : ';',
      q = iswin && isbash ? "'" : '"',
      r = (isbash || iszsh) ? '>|' : '>';
ping_cmd = `${p} echo "done" ${r}${q}${ipc_file}${q}`;
```
- For standard Windows command-line (CMD) the command separator is `&` – otherwise use `;`.
- For Windows BASH the path is wrapped in double-quotes.
- For all others the path is wrapped in single-quotes.
- For BASH or ZSH the redirect is `>|` otherwise it is just `>`.

thinkyhead, 2023-12-28:
The above code will be added to 2.1.51. If the issue isn't fixed in that release, open a new one.

closed 2023-12-28 by thinkyhead