MarlinFirmware/AutoBuildMarlin #51: USING MOTHERBOARD #if ENABLED() - AutoBuild FAILS TO SWITCH BOARDS

ftlpilot, 2021-10-07:
If you use MOTHERBOARD #if ENABLED() - AutoBuild FAILS TO SWITCH BOARDS - FYI, this is used in your supplied code for the Ender 5 (in configuration.h)

// ------------------------------------
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #if ENABLED(SKR_14_Turbo)
    #define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO
  #else
    #define MOTHERBOARD BOARD_MELZI_CREALITY
  #endif
#endif
// ------------------------------------

rigan0707, 2021-11-13:
H

ftlpilot, 2021-11-13:
To you - this is a feature request - to me - as a frustrated user - IT'S A BUG !

rongoodman303, 2021-12-16:
Yes, a new feature, but sorely needed.

**My Rationalization:**
- Being able to build for multiple variations of very similar printers from the same configuration files, but with different boards and/or custom machine names, allows all of those printers to share other common build attributes without having to have several different configuration files. 
- In my experience, it is really only necessary for two critical keyword cases that affect ability of ABM to successfully display the actual/intended configuration and do a build: MOTHERBOARD and CUSTOM_MACHINE_NAME, and only if there are other occurrences of those keywords in Configuration.h that are NOT commented out, indicating that they are controlled by #ifdef or other logic. 
- A good public example of where a printer manufacturer has provided problematic examples in the Marlin 2.0.9.2 examples collection, is the Alfawise U20. LGT, the manufacturer of this and other Alfawise and Longer printers, has provided an example that would never work with ABM for any of the other printers supposedly supported in that example, except the Alfawise U20 and its motherboard and custom machine name, because all of the others are later in the file and are surrounded by logic. So ABM simply scans and scrapes the first uncommented occurrence of what it is looking for.

**Proposed Short Term  Simple Solution:**
- To prevent ABM from attempting a build that might not be correct, continue scanning for multiple uncommented occurrences of those keywords, not just the first. If there is more than one uncommented occurrence, display an error message in the ABM panel of that situation, advise the developer to comment out options not needed, and not show or allow a build until it is corrected and it only finds one occurrence of each of those keywords. A link to some kind of Help explanation in that error message would also be helpful. 

**My current workaround:**
Build Configuration.h with a list of the available and supported options for MOTHERBOARD and CUSTOM_MACHINE_NAME, all commented out, instead of relying on logic selections. This requires that the firmware developer/builder manually pick and uncomment the one option they want for that build, and leave uncommented any that are not applicable. Can be tedious, but works for me until ABM has a better solution.

thinkyhead, 2023-12-28:
The parser is continually being extended to better parse `Configuration.h`, `Conditionals_LCD.h`, and `Configuration_adv.h` (in that order) for the benefit of the **Config Custom Editor** and the **Config Updater**. It's a challenging problem, but I'm making progress. I'll certainly endeavor to make the improved parser also apply to the build panel so that we can run the parser once at startup, then as-needed in response to configuration changes. It's just me working on this thing which is subordinate to Marlin Firmware itself, so please bear with the slow progress. I definitely want to have the **Config Updater** ready ahead of the next release, so will be focusing heavily on ABM as the next firmware release nears.

closed 2023-12-28 by thinkyhead