MarlinFirmware/AutoBuildMarlin #52: [BUG] Tronxy X5SA (Pro) needs _maple build to work

tommywienert, 2021-11-24:
Hello, i really hope this is the right place, as i did everything via ABM.
I got a new printer, Tronxy X5SA Pro and found the chitu firmware to be terrible, so i followed the guideline (https://github.com/MarlinFirmware/Configurations/tree/2.0.9.2/config/examples/Tronxy/X5SA) to build my own update.cbd to get Marlin onto it.
The Build succeeded, i put the update.cbd onto the SD card and turned the Power on, and got a series of two short beeps (5x) and then an "exitus" style never ending beep. I looked through everything and tried to change settings until i found a German Forum where several people told to use the _maple build instead of the normal one (https://drucktipps3d.de/forum/topic/tutorial-tronxy-x5sa-von-chitu-auf-marlin/?part=22#postid-173633). Stupid as i am i did not try that so far, but when i did, it worked first try. i changed _nothing_ in vs code (though i am using Codium) except to build the _maple instead of the normal build. Since i got some help on the Marlin Discord and gave feedback that it works now, i got told to create a bug ticket so that you guys know of this.
So ... what information do you need from me so that i can provide it?

Greetings
Tommy
[image]

thisiskeithb, 2021-11-24:
Please [create an issue in the main Marlin Firmware repository](https://github.com/MarlinFirmware/Marlin/issues/new?assignees=&labels=&template=bug_report.yml&title=%5BBUG%5D+%28bug+summary%29). This repo is for the Marlin AutoBuild extension for VSCode.

tommywienert, 2021-11-25:
Done

closed 2021-11-25 by tommywienert