MarlinFirmware/AutoBuildMarlin #61: Assertion error

orrinwitt, 2022-08-15:
I'm suddenly unable to upload firmware to my Archim2 board using the Auto Build Marlin extension in Visual Studio Code. I keep getting this "Assertion Error" which seems to suggest that PlatformIO isn't finding or correctly identifying my board (I'm guessing). 

I tried building and uploading the last Marlin that I knew worked, and still got the error, so it can't be something from the new version of the Marlin firmware. I also tried uninstalling and reinstalling the extensions, VSCode, and even tried it with a different computer that had never had it installed on it before (so it was a truly fresh install). Still got the same "Assertion error". 

Has something changed in either the Auto Build Marlin or the PlatformIO extensions that has broken something? Am I the only one having this issue?

See the output below. If you need something else to help diagnose the problem, let me know.

Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba, stlink
CURRENT: upload_protocol = sam-ba
Looking for upload port...
*** [upload] AssertionError : 
Traceback (most recent call last):
  File "C:\Users\Kara\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Action.py", line 1318, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "C:\Users\Kara\.platformio\platforms\atmelsam\builder\main.py", line 27, in BeforeUpload
    env.AutodetectUploadPort()
  File "C:\Users\Kara\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Util.py", line 737, in __call__
    return self.method(*nargs, **kwargs)
  File "C:\Users\Kara\.platformio\penv\lib\site-packages\platformio\builder\tools\pioupload.py", line 113, in AutodetectUploadPort
    UPLOAD_PORT=SerialPortFinder(
  File "C:\Users\Kara\.platformio\penv\lib\site-packages\platformio\device\finder.py", line 144, in find
    device = self._find_known_device()
  File "C:\Users\Kara\.platformio\penv\lib\site-packages\platformio\device\finder.py", line 221, in _find_known_device
    for board_config in p.get_boards().values():
  File "C:\Users\Kara\.platformio\platforms\atmelsam\platform.py", line 105, in get_boards
    result[key] = self._add_default_debug_tools(result[key])
  File "C:\Users\Kara\.platformio\platforms\atmelsam\platform.py", line 154, in _add_default_debug_tools
    assert openocd_chipname
AssertionError
=========================================================== [FAILED] Took 86.65 seconds =========================================================== 

Environment    Status    Duration
-------------  --------  ------------
DUE_archim     FAILED    00:01:26.652

orrinwitt, 2022-08-15:
Update: This must be a platformio issue. I tried using the marlin firmware updater plugin for Octoprint on my Rpi, and it threw the same error. Still not sure what to make of it though or where to even start to diagnose how to fix it.

here's the output from the aforementioned plugin:

Processing DUE_archim (board: marlin_archim; platform: atmelsam; framework: arduino)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-bossac @ ~1.10700.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking
Tool Manager: tool-bossac@1.10700.190624 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/marlin_archim.html
PLATFORM: Atmel SAM (8.1.0) > Archim
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduino-sam @ 1.6.12 
 - framework-cmsis @ 1.40500.0 (4.5.0) 
 - framework-cmsis-atmel @ 1.2.2 
 - tool-bossac @ 1.10700.190624 (1.7.0) 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SPI @ 1.0
|-- Wire @ 1.0
|-- SPI @ 1.0
Building in release mode
Compiling .pio/build/DUE_archim/src/src/inc/Warnings.cpp.o
Checking size .pio/build/DUE_archim/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.0% (used 8884 bytes from 98304 bytes)
Flash: [====      ]  35.9% (used 188220 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba, stlink
CURRENT: upload_protocol = sam-ba
Looking for upload port...

Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/en/latest/core/installation/udev-rules.html

*** [upload] AssertionError : 
Traceback (most recent call last):
  File "/home/pi/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Action.py", line 1318, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "/home/pi/.platformio/platforms/atmelsam/builder/main.py", line 27, in BeforeUpload
    env.AutodetectUploadPort()
  File "/home/pi/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737, in __call__
    return self.method(*nargs, **kwargs)
  File "/home/pi/.octoprint/data/marlin_flasher/platformio/lib/python3.7/site-packages/platformio/builder/tools/pioupload.py", line 118, in AutodetectUploadPort
    ).find(initial_port)
  File "/home/pi/.octoprint/data/marlin_flasher/platformio/lib/python3.7/site-packages/platformio/device/finder.py", line 144, in find
    device = self._find_known_device()
  File "/home/pi/.octoprint/data/marlin_flasher/platformio/lib/python3.7/site-packages/platformio/device/finder.py", line 221, in _find_known_device
    for board_config in p.get_boards().values():
  File "/home/pi/.platformio/platforms/atmelsam/platform.py", line 105, in get_boards
    result[key] = self._add_default_debug_tools(result[key])
  File "/home/pi/.platformio/platforms/atmelsam/platform.py", line 154, in _add_default_debug_tools
    assert openocd_chipname
AssertionError
========================= [FAILED] Took 49.22 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
DUE_archim     FAILED    00:00:49.221
==================== 1 failed, 0 succeeded in 00:00:49.221 ====================

orrinwitt, 2022-08-15:
and after installing the "99-platformio-udev.rules" as instructed via the link above, I tried again and still got this:

Processing DUE_archim (board: marlin_archim; platform: atmelsam; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/marlin_archim.html
PLATFORM: Atmel SAM (8.1.0) > Archim
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduino-sam @ 1.6.12 
 - framework-cmsis @ 1.40500.0 (4.5.0) 
 - framework-cmsis-atmel @ 1.2.2 
 - tool-bossac @ 1.10700.190624 (1.7.0) 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SPI @ 1.0
|-- Wire @ 1.0
|-- SPI @ 1.0
Building in release mode
Compiling .pio/build/DUE_archim/src/src/inc/Warnings.cpp.o
Checking size .pio/build/DUE_archim/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.0% (used 8884 bytes from 98304 bytes)
Flash: [====      ]  35.9% (used 188220 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba, stlink
CURRENT: upload_protocol = sam-ba
Looking for upload port...
*** [upload] AssertionError : 
Traceback (most recent call last):
  File "/home/pi/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Action.py", line 1318, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "/home/pi/.platformio/platforms/atmelsam/builder/main.py", line 27, in BeforeUpload
    env.AutodetectUploadPort()
  File "/home/pi/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737, in __call__
    return self.method(*nargs, **kwargs)
  File "/home/pi/.octoprint/data/marlin_flasher/platformio/lib/python3.7/site-packages/platformio/builder/tools/pioupload.py", line 118, in AutodetectUploadPort
    ).find(initial_port)
  File "/home/pi/.octoprint/data/marlin_flasher/platformio/lib/python3.7/site-packages/platformio/device/finder.py", line 144, in find
    device = self._find_known_device()
  File "/home/pi/.octoprint/data/marlin_flasher/platformio/lib/python3.7/site-packages/platformio/device/finder.py", line 221, in _find_known_device
    for board_config in p.get_boards().values():
  File "/home/pi/.platformio/platforms/atmelsam/platform.py", line 105, in get_boards
    result[key] = self._add_default_debug_tools(result[key])
  File "/home/pi/.platformio/platforms/atmelsam/platform.py", line 154, in _add_default_debug_tools
    assert openocd_chipname
AssertionError
========================= [FAILED] Took 37.09 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
DUE_archim     FAILED    00:00:37.091
==================== 1 failed, 0 succeeded in 00:00:37.091 ====================

thinkyhead, 2022-10-23:
You'll need to update your gcc installation to version 12 following the instructions in the latest `native.ini` in `bugfix-2.1.x`.

orrinwitt, 2022-10-25:
I know this issue is closed, but I'm not sure how to do that... And I'm still getting this error no matter what I do. Even with different computers.

orrinwitt, 2022-10-28:
Can I please get some help with this?? I've completely exhausted everything I know to do, including starting from scratch with a new firmware build (freshly unzipped). I have learned stuff along the way, but nothing that has made any difference. Still getting this error:

`*** [upload] AssertionError : 
Traceback (most recent call last):
  File "C:\Users\orrin\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Action.py", line 1318, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "C:\Users\orrin\.platformio\platforms\atmelsam\builder\main.py", line 27, in BeforeUpload
    env.AutodetectUploadPort()
  File "C:\Users\orrin\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Util.py", line 737, in __call__
    return self.method(*nargs, **kwargs)
  File "C:\Users\orrin\.platformio\penv\lib\site-packages\platformio\builder\tools\pioupload.py", line 113, in AutodetectUploadPort
    UPLOAD_PORT=SerialPortFinder(
  File "C:\Users\orrin\.platformio\penv\lib\site-packages\platformio\device\finder.py", line 144, in find
    device = self._find_known_device()
    for board_config in p.get_boards().values():
  File "C:\Users\orrin\.platformio\platforms\atmelsam\platform.py", line 105, in get_boards
    result[key] = self._add_default_debug_tools(result[key])
  File "C:\Users\orrin\.platformio\platforms\atmelsam\platform.py", line 154, in _add_default_debug_tools
    assert openocd_chipname
AssertionError`

thinkyhead, 2023-12-28:
This would've been an issue with Marlin INI settings or more likely with PlatformIO, and not with ABM which simply invokes the PlatformIO build. If this is still a problem please file the issue over at the main project.

closed 2023-12-28 by thinkyhead