MarlinFirmware/AutoBuildMarlin #65: Cannot install on Linux

kurthuwig, 2023-02-27:
When search for "Marlin" in the extensions, I get only these results:
* Mario: The Block J...
* Star Rod
* nightly
* 
[image]

thisiskeithb, 2023-02-27:
Are you using [VSCodium](https://github.com/VSCodium/vscodium) by chance? VSCode’s Marketplace isn’t used in those builds due to licensing issues, so it won’t show.

You could try to install it manually: https://marketplace.visualstudio.com/items?itemName=MarlinFirmware.auto-build

kurthuwig, 2023-02-27:
I use this version:

1 community/code 1.75.1-1 (14.3 MiB 102.8 MiB) (Installed)
    The Open Source build of Visual Studio Code (vscode) editor

Version: 1.75.1
Commit: 441438abd1ac652551dbe4d408dfcec8a499b8bf
Date: 2023-02-09T20:16:38.064Z
Electron: 19.1.9
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 6.1.12-1-MANJARO
Sandboxed: No

The manual installation worked fine, thanks!

thisiskeithb, 2023-02-27:
> I use this version:
> 1 community/code 1.75.1-1 (14.3 MiB 102.8 MiB) (Installed)
> The Open Source build of Visual Studio Code (vscode) editor

Ahh. It has the same extension licensing restrictions since it’s not the official VSCode build from Microsoft.

kurthuwig, 2023-02-27:
Ok, but why do I see other extensions that I can install? Which licensing restriction apply? I thought Marlin were GPL?!

thisiskeithb, 2023-02-27:
> I thought Marlin were GPL?! 

It is, but the restriction is from Microsoft, not Marlin.

The [VSCodium readme on extensions/marketplace](https://github.com/VSCodium/vscodium/blob/master/DOCS.md#extensions--marketplace) covers this better than I can explain here, but the short version is:

> Unfortunately, as Microsoft https://github.com/microsoft/vscode/issues/31168 or redistribution of .vsix files from it, in order to use VS Code extensions in non-Microsoft products those need to be installed differently.

So any non-Microsoft build of VSCode is forced to use a different extension marketplace (like https://open-vsx.org/ or whatever your fork of VSCode uses), or they would be in violation of the licensing.

kurthuwig, 2023-02-28:
Thanks for the explanation!

closed 2023-02-28 by kurthuwig