MarlinFirmware/AutoBuildMarlin #66: Run command fails in Windows PowerShell

lukasradek, 2023-04-07:
I am using Windows 11, VSCode, PIO, Auto Build Marlin to build simulator.
It builds fine but it fails to Run with the following error.

[image]

It executes this command `START /b C:_\Lukas\Programming\marlin\Marlin.pio\build\simulator_windows\debug\MarlinSimulator.exe`.
I have discoved that the culprit is probably the /B flag in combination with Windows PowerShell which is now default on Windows. Running the command in cmd.exe works. Running the commnad in PowerShell **without** /B works.

As a workaround, it is possible to switch default terminal app in VSCode to external and in Windows Terminal set "Command line" as default terminal app.

thinkyhead, 2023-08-07:
I'm not sure how we can detect PowerShell as the default system shell, unless that can be found as a PlatformIO / VSCode setting. If it can't be done automatically then I'll need to add a new Settings option.

lukasradek, 2023-08-07:
Me neither. Is it possible to remove the /B flag?

thinkyhead, 2023-10-05:
Maybe it can be run like so in both the old and new shells:
```powershell
cmd.exe /c START /b C:_\Lukas\Programming\marlin\Marlin.pio\build\simulator_windows\debug\MarlinSimulator.exe
```

thinkyhead, 2023-12-28:
ABM 2.1.51 will be adjusting commands based on shell type. If this is still a problem after that release then please file a new report.

closed 2023-12-28 by thinkyhead