MarlinFirmware/AutoBuildMarlin #69: Can't compile for BTT SKR MINI E3 V3.0

mfraser, 2023-05-11:
When trying to compile firmware for a BTT SKR MINI E3 V3.0 with STM32G0B1RE processor, Auto Build Marlin tries and fails to download https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip with:
`Error: Got the unrecognized status code '401' when downloaded https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip`

thisiskeithb, 2023-05-11:
GitHub is having issues: https://www.githubstatus.com/

mfraser, 2023-05-11:
> GitHub is having issues: https://www.githubstatus.com/

Still not able to download and looking at the status page all is OK.

thisiskeithb, 2023-05-11:
> Still not able to download and looking at the status page all is OK.

You probably caught it at a bad time. It works fine now:

<details><summary>build details:</summary>
<p>

```prolog
Processing STM32G0B1RE_btt (platform: ststm32@~14.1.0; board: marlin_STM32G0B1RE; framework: arduino)
----------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
Unpacking  [####################################]  100%          
Tool Manager: toolchain-gccarmnoneeabi@1.90201.191206 has been installed!
Tool Manager: Installing platformio/framework-cmsis @ ~2.50700.0
Unpacking  [####################################]  100%          
Tool Manager: framework-cmsis@2.50700.210515 has been installed!
Tool Manager: Installing https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoststm32@0.0.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32G0B1RE.html
PLATFORM: ST STM32 (14.1.0) > STM32G0B1RE
HARDWARE: STM32G0B1RET6 64MHz, 144KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-arduinoststm32 @ 0.0.0 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|-- U8glib-HAL @ 0.5.2
|-- SlowSoftI2CMaster @ 1.0.0
|-- SlowSoftWire @ 0.0.0+20220909150431
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0.0
|-- Wire @ 1.0.0
|-- Servo @ 1.1.2
Building in release mode
Compiling .pio/build/STM32G0B1RE_btt/FrameworkArduinoVariant/PeripheralPins.c.o
[snip]
```

</p>
</details>

mfraser, 2023-05-12:
I must be missing something as I'm still getting the error:
```
platformio run -e STM32G0B1RE_btt ; echo "done" >|"/tmp/ipc"
mfraser@Rachael:~/Marlin/Ender3V2S1$ platformio run -e STM32G0B1RE_btt ; echo "done" >|"/tmp/ipc"
Processing STM32G0B1RE_btt (platform: ststm32@~14.1.0; board: marlin_STM32G0B1RE; framework: arduino)
-------------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
Error: Got the unrecognized status code '401' when downloaded https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
```

I've got 3 extensions installed Auto Build Marlin, C/C++ & PlatformIO IDE. This is on Linux x64.

mfraser, 2023-05-13:
All working now, copied the .platformio directory from another install it it worked.

closed 2023-05-13 by mfraser