MarlinFirmware/AutoBuildMarlin #72: Publish extension to open-vsx.org

qwertychouskie, 2023-05-29:
Hi!

I'm a [VS Codium](https://github.com/VSCodium/vscodium) user, a FLOSS version of VS Code, it's the same code but without the Microsoft's telemetry and branding.

I want to request if possible to add the PlatformIO IDE extension on the [open-vsx.org](https://open-vsx.org/) market so that we don't need to use with the Microsoft marketplace, as it is unclear if it's legal to use it on VS Code forks, see https://github.com/microsoft/vscode/issues/31168.

This is [how to publish a new extension](https://github.com/eclipse/openvsx/wiki/Publishing-Extensions).

Thanks!

(A temporary workaround is to manually download the extension from https://marketplace.visualstudio.com/items?itemName=MarlinFirmware.auto-build)

thinkyhead, 2023-08-07:
Thanks. I will look at this for the next update. Pretty busy with the next Marlin release lately.

qwertychouskie, 2023-08-07:
Awesome!  Let me know if there's anything I can do to help.

qwertychouskie, 2023-08-07:
I should also note that the PlatformIO extension [has not yet been published to open-vsx](https://github.com/platformio/platformio-vscode-ide/issues/1802#issuecomment-1567465212).  I still think it's beneficial to add AutoBuildMarlin to open-vsx though, as it's one fewer extension people like me have to manually download.  You might want to put a note in the description on open-vsx though.

marek22k, 2023-08-14:
I would also be happy if it was available in VSCodium!

thisiskeithb, 2023-08-14:
> I should also note that the PlatformIO extension https://github.com/platformio/platformio-vscode-ide/issues/1802

This is due to using Microsoft extensions / products outside of VSCode which goes against Microsoft’s terms of service/license.

Let me preface that I was also on the VSCodium train until I ended up having to deal with tons of missing extensions & potential license/terms of service violations.

Since Marlin relies heavily on PlatformIO, publishing ABM to vsx.org would be premature until the full plugin/extensions/tool chains can be published over there.

closed 2023-12-28 by thinkyhead