MarlinFirmware/AutoBuildMarlin #76: New version of AutoBuildMarlin fails due to no project loaded into workspace

nicwilson58, 2023-11-20:
My AutoBuild Marlin was working fine on V2.1.47,   yesterday it updated itself to V2.1.49 and since then it no longer functions as it states there is no project loaded into the workspace.    I rolled back to V2.1.47 and it was immediately fixed,  so I cannot see it being anything else as its all been very stable for a long time.     I did try deleting the .platformio folder from the home directory but this has no effect.  The only way I have found to fix it is to use V2.1.47

kainmo, 2023-11-21:
I have the same issue with V2.1.49，and I roll it back to V2.1.47，it is worked again，thanks，the new version has a stable problem obviously.

nicwilson58, 2023-12-07:
So 2 weeks later and no input on this issue?  I just re-tested this issue across 15 PC's and each had this exact same issue,  and every one was fixed by rolling nack to  2.1.47

thinkyhead, 2023-12-12:
I could go hunting around at random hoping to find the broken line or misplaced semicolon, but it would be very helpful if someone who is seeing the issue could provide some configs that cause the issue to manifest. I have not seen this issue lately while working on the **Config Migration** feature.

thinkyhead, 2023-12-12:
> So 2 weeks later and no input on this issue?

@nicwilson58

Nic, I've already told you once that there is no need to be rude and impatient. Marlin is a very busy project with very limited resources, and I have a huge pile of work each day that I have to prioritize, not leaving much time for a life. If you cannot be empathetic, patient, and understanding then … I dunno, man. You really should try to be nicer.

nicwilson58, 2023-12-12:
If simply asking for an update you find rude,  then forget it,  the only person being rude here is you.   Just forget the whole thing.
Re-read what you said,  and what 2 of us said, then tell me who is rude

nic

nicwilson58, 2023-12-12:
a suggestion for you, instead of sending out rude and nasty email,ls,  try updating the thread  no one minds how long it takes,  but ignoring the thread 100% means no one knows if anything is occurring or if the issue has been confirmed by more than two commentors.

kainmo, 2023-12-12:
Be peaceful man, find the problem and fixed it is all we need, not fighting, we all hope the marlin project go far and smooth, calm down man. My marlin version is 02010201 with LPC1768 board（mks sgenl v1），using the vscode（1.84.2），and I found that the AutoBuild Marlin V2.1.48 and 2.1.49 has the same issue， the latest working version is V2.1.47. I am an mechanism engineer, I have no ideas with the codes,,,,,,,,so please help.

thinkyhead, 2023-12-12:
@nicwilson58 — My apologies, Nic. I may be over-sensitive because we haven't spent near enough time conversing and getting to understand one another. If you were "simply asking" for an update, clearly I just don't understand your style and sense of humor. I may also be over-sensitive to you in particular due to our limited interactions. I will try to be more careful interpreting your statements in future

Meanwhile, please understand that for many busy open source projects, two weeks is not an uncommon period of time to await a response. Contributors to open source projects do it for fun and in their spare time and may even take time off to do other things. While awaiting a reply, the best thing you can do is to add more information to your issue to help the person(s) who will eventually help you solve the problem.

Thank you for your understanding. We're all in this together.

thinkyhead, 2023-12-28:
> the latest working version is V2.1.47.

Still cannot reproduce. If someone would ZIP up and drop their failing `Configuration.h` and `Configuration_adv.h` files into a reply in this thread, they will help me to see where the problem lies.

houseofbugs, 2023-12-28:
You know all the comments from @nicwilson58 here and zero logs or even screenshots of the error. How can someone troubleshoot the error with no details other than "It DoEsNt WoRk".

thinkyhead, 2023-12-28:
> How can someone troubleshoot the error

Sometimes if I stare at the code hard enough the bug will stand out. Unfortunately, I can't discern any obvious issue with the changes from 2.1.48, so it really will help to have those configs to test with. A reliable config parser that can't be broken by a stray backslash will be very important going forward, especially for the **Config Updater**.

But no rush, I know everyone is busy with other things during this time of the year.

thinkyhead, 2023-12-28:
Okay, I got some help from #77 which pointed out that BlueR configs trigger the bug. Is it the same bug? Maybe, probably, sure, but we will see!

>> When configuration.h has a board option (MK13) like Bluer V3, the autoconfiguration not load."

> Thank you for the specifics. That's the exact information I was looking for! I grabbed the BlueR configs with use_example_configs "Two Trees/BlueR/BlueR V1" and sure enough ABM doesn't like them. Once I track down the exact issue, it may be a quick fix, or it might require a bigger update. (At the moment the extension is just scraping the configs to get a basic summary. Once I switch over to using the ConfigSchema class the configs will be parsed in a more complete way.)

thinkyhead, 2023-12-28:
Aha, well that (#77) was an easy one. The error shown in the Javascript console of VSCode was "cannot read property ('comment') of undefined" at `schema.js:818`. The parser just needs to handle the unexpected case of an "end-of-line" comment before any `#define` lines. So this one line above 818 solves that particular issue:
```js
if (last_added_ref === undefined) continue;
```
Will this fix _your_ exact issue? We shall see….

kainmo, 2023-12-28:
V2.1.51 is worked again，thanks everyone！

nicwilson58, 2023-12-28:
Confirmed fixed on all the PC's farm.  Many thanks.

> @nicwilson58 — My apologies, Nic. I may be over-sensitive because we haven't spent near enough time conversing and getting to understand one another. If you were "simply asking" for an update, [...]

My Aplogies too, my sense of humor is usually a bit dry.   As a programmer I understand your need for info,  but on a small project like this I simply to a highlighted comparison of changes, which are usually minimal and easier to spot what broke it.  I cannot see any logs,  and as the window never opened,  its hard to provide any more informaiton on this particular error.   On such errors in my team I request specific info in Jira if I require more info than has been provided,  I never expect a user to know what info to provide.     Thanks for the fix anyway.

thinkyhead, 2023-12-29:
> Confirmed fixed on all the PC's farm. Many thanks.

I'm thrilled to hear that the comment thing was the problem. I'm putting a lot of stake in this schema parser for a robust configuration editor and especially to automate config update. That's one of the biggest hassles when it comes to updating to the next version. If I can get it built into the actual build process — with or without ABM — that will be even better.

Much appreciation for your followup. This is one of the most interesting and fun projects I have ever been involved with, and my first experience being a project lead and maintainer on something this expansive. I try to keep to the technical details but of course we are all humans with our good and bad days. It's been 12 full months since the last release and I want to get everything "just right" for this next one (2.1.3 … or maybe better, "2.2.0"), and I want to especially help simplify things for vendors and maintainers of Marlin distros. Please let us know if there is anything we can do to improve the config or build system or this VSCode extension that would make your job easier.

closed 2023-12-29 by thinkyhead