MarlinFirmware/AutoBuildMarlin #77: AutoBuild fails to open a configuration file with board options like Two Trees Bluer V3

espivetta, 2023-12-14:
When configuration.h has a board option (MK13) like Bluer V3, the autoconfiguration not load.

thisiskeithb, 2023-12-14:
Duplicate of https://github.com/MarlinFirmware/AutoBuildMarlin/issues/51

thinkyhead, 2023-12-28:
Thank you for the specifics. That's the exact information I was looking for! I grabbed the BlueR configs with `use_example_configs "Two Trees/BlueR/BlueR V1"` and sure enough ABM doesn't like them. Once I track down the exact issue, it may be a quick fix, or it might require a bigger update. (At the moment the extension is just scraping the configs to get a basic summary. Once I switch over to using the `ConfigSchema` class the configs will be parsed in a more complete way.)

Anyway, if more discussion is needed we can do that over at issue #51.

closed 2023-12-28 by thinkyhead