MarlinFirmware/AutoBuildMarlin #78: Configuration Tool

brunofresco, 2024-01-11:
Will the configuration tool still be updated in the future? I really like Marlin Auto Build and id like to see it get better with time :)

thinkyhead, 2024-01-13:
As time allows. My priority is currently on the next Marlin release, then for ABM my emphasis is on the Config Editor (which is pretty cool) and the config migration tool, which is in progress. The development of the Config Custom Editor was a useful thing to drive the development of the Config Schema Parser and other important pieces that will come into play in a more thought-out and designed config editor interface. It's only recently that the schema parser is not exploding every two seconds, so things are starting to fall into place.

Config editing ties in with config documentation, and that's another thing that a working schema parser should help a lot going forward once we can get well organized YAML exports out of it.

The Geometry Panel has been on my mind lately. Here's a peek at what's been happening in Inkscape…

[image]

brunofresco, 2024-01-15:
Thanks for the reply, that looks cool and im very happy to know what you are planning.
I will support you on Patreon, I really appreciate your work.