MarlinFirmware/AutoBuildMarlin #92: reveal_env_build uses wrong parameters for Windows Explorer

TheAndr0id, 2025-05-23:
First up, I'm using Windows 7.  Why?  Because it's an old laptop that newer Windows won't.

The build works fine.  The binaries compile (I haven't tried to update the firmware yet) but the compile finishes with no unexpected errors or warnings.

The "Auto Reveal Build" function just opens an Explorer window to my default home directory, not the directory of the .bin file (it is there).

Looking at the reveal_env_build function, the code:

```
switch (process.platform) {
    case 'win32': cmd = `Explorer /select,${stat.filename}`; break;
    case 'darwin': cmd = `open -R ${stat.filename}`; break;
    default: cmd = '`which xdg-open open other-open | grep -v found | head -n1` .';
  }
```

Those options to Explorer are not correct...  (...and that's why I get a Explorer window to my home directory)

I've disabled the Auto Reveal function.

ellensp, 2025-08-15:
I cannot replicate this issue

The  options to Explorer are correct.
Explorer → launches File Explorer
/select, → tells it to open the folder and highlight the given file
${stat.filename} -> is just the file name  eg firmware.hex. firmware.bin 

It should really have quotes around the filename  `case 'win32': cmd = `Explorer /select,"${stat.filename}"`; break;`

The directory is set with  `command_with_ping(aterm, `cd "${escpath}"`);`

TheAndr0id, 2025-08-17:
There's no need to get hostile.

I made it pretty clear I don't know your code - and don't really want to - I'm a Unix guy forced to use an old Windows box in the short term.  The command_with_ping() thing didn't return any google hits to help me know what it does, and I was too tired last night to go digging through code to find out.

The bottom line is it currently doesn't work under Windows 7.  I've shown the simple fix for the problem, fix it or don't.  It's your software, weather you want to improve it is up to you.

I already have a work around.  When I'm back to my regular environment I'll return to using make from the command line.

TheAndr0id, 2025-08-19:
I actually just added the two quotes, works fine now.... :)

thinkyhead, 2025-10-02:
> Every development always states they want input and want users to report issues. Really? Are you sure?

See, guys! This is what happens, and it ends up living in the collective memory forever. ABM is as much my personal pet project as it is a part of Marlin, and I put a lot of time and effort and love into improving it. If you come along and insult people —totally against the spirit of open source volunteerism— it doesn't just go to your own reputation, it ends up impugning me and harming my reputation, as well.

I will not tolerate rudeness any more. If I see it again, contributors will be banned. I don't care how productive and helpful you are. This is a red line.

closed 2025-10-02 by thinkyhead