MarlinFirmware/AutoBuildMarlin #93: Firmware.bin uses wrong time stamp in M115 (make/dependencies update)

TheAndr0id, 2025-06-28:
When doing a non-clean build (aka, only compiling the files that have changed) the new firmware binary has the time stamp from the initial build, not the latest build. 

The make dependencies  for `Marlin//src/gcode/host/M115.cpp`  need to be set to always.

firmware-20250627-224300.bin loaded (and confirmed - notable changes unique to this version)
```
M115
FIRMWARE_NAME:Marlin 2.1.2.5 (Jun 27 2025 14:46:19) ....
```

Which is a timestamp of a previous firmware build.

Did a clean build and M115 reports the correct timestamp.

TheAndr0id, 2025-08-19:
Thanks for looking into this.  My current work around is to remove the M115.o object file from the build tree before I compile so VSCode will be forced to rebuild it.  On this old laptop I'm currently stuck using a full rebuild can take more than 30 minutes.

I'm a Unix guy.  In my world we (generally) use `make` to manage code dependencies.  In make there are options to force an object file to always be recompiled no matter if it's changed.  In old school make files (eg: SunOS or HP/UX) we specified a terminal device file as a dependency since they always change.  Yeah, it was a hack, but it worked...

Sorry, I didn't realize this would be this size of job.

classicrocker883, 2025-08-19:
those are good skills to have, just seems so much easier when you have full control, like in terminal.  
I actually updated the whole implementation, make a **PR** over on Marlin repo, you may have noticed the mention right above your comment.

i know before it seemed messy having a separate header file, but i found an easier way

closed 2025-10-02 by thinkyhead